Robo3D/Marlin #1: Do not allow heated bed to power on while thermistor is not connected.

sonagit, 2016-12-27:


AllenMcAfee, 2016-12-27:
We need to verify a safe method of sensing the presence of the heated bed in RoboOS either through the thermistor reading or a simple circuit to a Digital pin on the Mainboard.  We are assuming the thermistor reading is the first method to try.

The assumption here is that Marlin does have temperature limits that will give Max/Min errors, but does not prevent these errors by sensing if the bed is present.  What we want to do is give a print command that includes a heated bed value, and have RoboOS recognize if the bed is present or not.  

What currently happens is the heated bed command is given at the start of a print, and Marlin will disconnect the printer due to a heater error (MIN or MAX temp error) if the bed is not present.  

What we want to see happen is a window similar to Zoffset that warns user the bed is not detected prior to a print, regardless of the target temp command in the gcode being present or not.

This ability to sense the bed should not be dependent on the heated bed command, but rather on a temperature to be present at all.  If the temp is out of range, we can introduce an appropriate warning code and window (since this indicates the bed is broken somehow).

AllenMcAfee, 2016-12-27:
This should also be driven by the printer profile in Octoprint (heated bed option) so it can be a universal part of RoboOS, rather than just in the R2 version.

closed 2017-04-03 by AllenMcAfee