Robo3D/Marlin #17: Check PID settings.

Ximidar, 2017-11-10:
PID is failing on some printers running 240 or greater temperatures. Run a PID autotune to see if it is just those printers, or if it is all printers.

Ximidar, 2017-11-20:
I ran the PID tuning on my machine for bed and for extruder 1

I set the target temp for extruder 1 to 280 My PID came out as: 

PID | Original | Updated
------------ | ------------- | -------------
P | 23.80 | 20.37 
I | 1.70 | 1.20
D | 85.00 | 86.78

For the bed I set the target temp to 100 My PID came out as: 

PID | Original | Updated
------------ | ------------- | -------------
P | 107.41 | 242.41 
I | 21.29 | 41.10
D | 135.50 | 357.47

AllenMcAfee, 2017-11-20:
@Ximidar how many times have you run each of these tuning tests?  We will want to run each at least 3 times and find a mean value if possible.

Ximidar, 2017-11-20:
@AllenMcAfee I only ran each test twice, What Temperature target points do you use for each heater?

closed 2017-12-20 by Ximidar