Robo3D/Marlin #18: [FR] Change leveling scheme to bilinear

WheresWaldo, 2017-11-19:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

Linear leveling assumes that your bed is flat and only tilted in one major axis. If for some reason you are not mechanically level to the best of your ability you may still see leveling issues. The reason is that linear leveling takes all its measurement and tries to fit a best plane that encompasses all the points measured, AT THE SAME TIME. So you end up with a best fit that does not truly represent the actual topography of the bed surface. 

Bilinear leveling on the other had is designed to create a plane between every four points on the bed, you end up with multiple planes that do follow the contour of the bed much more closely (limited by the number of points you probe).

For the manufacturer and the end user there is no difference in what either sees while leveling the bed. Marlin will do all the same calculations in both with the difference being that for Bilinear it does it multiple times. The end result is better first layer adhesion and less fiddling with compromises trying to fix this with Home or Extruder Offsets.
------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

AllenMcAfee, 2017-11-20:
@WheresWaldo I fully agree with your assessment.  We started testing Bilinear leveling last week using the new G35 and G36 commands defined in the RC branch.  Somehow we are handling the offset incorrectly in the Z-offset wizard as the testing led to air prints.  We should be able to figure this soon enough.

ABL has tested really well using the new G36 scheme to adjust the probe offset.  I'm sure it'll work just as well with Bilinear once we tweak our Octoprint wizards.

AllenMcAfee, 2017-11-22:
@WheresWaldo here is the branch where we are working on Bilinear.  Testing these last 2 days have gone decently well, we are just ironing out a few small details.  This will work with the version of Octoprint you have on your R2, just be sure to replace G28 and G29 with G36.  

You will have to set the start of your priming line to X10.  This helps overcome the issue we are seeing in #10

WheresWaldo, 2017-11-22:
I eliminated the prime line since I print everything with a skirt. So I am not seeing issue 10 at all. 

Bilinear should work exactly like linear since the output is basically the same. I am not surprised that you are finding no issues switching. It also can actually handle warped beds where linear cannot.

AllenMcAfee, 2018-02-13:
Documenting this for future reference since some users reported errors who's cause were not immediately obvious:

**Issue:** Bilinear leveling not compensating correctly - appears to compensate on one side of the bed correctly, but not on the opposing side (usually front to back).  Prints too high on one side.

 **Symptoms:** The symptom will present itself as correctly printing on the front end of the bed (assumingely because that's where the offset is set), but not printing correctly in the back.  This incorrect compensation can manifest as printing too far on one side of the bed, or too close depending on the bed's slope.

**Cause** ENABLE_LEVELING_FADE_HEIGHT is activated in firmware, but `M420 Z<height>` or `M420 S<bool>` is not set in the START GCODE in Octoprint or within the .GCODE print file itself.  Since ENABLE_LEVELING_FADE_HEIGHT requires a compensation distance to be set using `M420 Z<height>`, the firmware defaults this to immediately be set at 0mm (the start of the print), effectively canceling out Z compensation movements in the initial print layer. 

**Solution** A few solutions are available:
1. Set a height in mm for M420, such as `M420 Z30`
2. Turn off ENABLE_LEVELING_FADE_HEIGHT in GCODE using `M420 S0`
3. Or deactivate ENABLE_LEVELING_FADE_HEIGHT at Z in Configuration.h if the function will not be used.

**From Marlin documentation**
   *   S[bool]   Turns leveling on or off
   *   Z[height] Sets the Z fade height (0 or none to disable)
   *   V[bool]   Verbose - Print the leveling grid

WheresWaldo, 2018-02-13:
ENABLE_LEVELING_FADE_HEIGHT has been included in Marlin since at the very least December 2016. I just looked at the main Marlin GITHUB and see one currently open. There were other issues, most fixed by the time 1.1.7 was released. The issue that looks like this one is https://github.com/MarlinFirmware/Marlin/issues/9529 it would be good if you chimed in with what you have found. I do not use BILINEAR, as you know, but my understanding is that if you are going to level in its simplest form BILINEAR handles slight bed warps better than LINEAR.

WheresWaldo, 2018-02-16:
Doing more research on this issue, I have been reading a lot of open and closed issues on the main Marlin GITHUB. It may already be fixed, even if inadvertently, in Marlin 1.1.8. Consider making 1.1.8 an RC priority immediately after you complete work on the current RC.

AllenMcAfee, 2018-02-22:
@WheresWaldo Thanks for that.  I've posted my findings to the issue.. let's see what they say.  There are users in there reporting the same issue in 1.1.8, and I can concur with their findings myself.  The same issue was presented in 1.1.8 with `ENABLE_LEVELING_FADE_HEIGHT` active and M420 Z not handled.  

The only thing that is holding me back from signing off 1.1.8 is the way the cancel print function is now handled.  The immediate deactivation of the steppers leads to a very long pause before the cancel GCODE is executed.  If the nozzle is on the bed or in another bad spot, the issue that caused the cancel is more likely to be exaggerated.  Are you seeing the same cancel behavior?

WheresWaldo, 2018-02-22:
I found that pause issue also on cancelling. On a Cancel from the LCD it takes quite a while before the OctoPrint script moves the head. There is also a slight delay in executing the print completion script of about 2 seconds. I resorted to adding a lift at the end of my slicer produced G-Code. I am not sure that the same delay happens if you cancel a print directly from OctoPrint but I will check that out sometime today. 

If it is only through the LCD then I have no place to report an issue and no way to track it afterwards.

WheresWaldo, 2018-02-22:
Did this several times, with stopwatch in hand. Canceling a print from LCD takes just over 5 seconds to return control to OctoPrint's Cancel Script. If you cancel a print directly from the web interface of OctoPrint then it takes less than 2 seconds before the Cancel Script runs. So I don't believe this is a Marlin 1.1.8 issue. My printer is currently running as generic a copy of Marlin 1.1.8 as the Robo hardware allows.

AllenMcAfee, 2018-09-27:
@WheresWaldo closing this out as we rolled in bilinear in March, and the pause at the end of the print has been traced back to Octoprint on the C2.  Thanks for your help!

closed 2018-09-27 by AllenMcAfee