Robo3D/Marlin #19: 1.1.9 -  LEDs turn off when idle.

mnacinopa, 2017-11-21:
### Bug Report
- **Description:**  Marlin 1.1.9 is shutting off the LEDs when idle. 
- **Expected behaviour:** LEDs to remain on at all times. 
- **Actual behaviour:** LEDs are active when a print is running, or when a motor control is executed. 
- S**teps to reproduce:** Let the printer stay idle ~2minutes.

Ximidar, 2017-11-22:
Which LEDs are you talking about? The Blue LEDs will only be on when Temperature is turned on, Then the other LEDs have specific functions. Most LEDs should be turned off while idle as power is not running to the heaters.

AllenMcAfee, 2018-02-13:
This was found to be an issue with bad LEDs received from the factory.  No issue was found in the firmware, or with the Mainboard after testing.

closed 2018-02-13 by AllenMcAfee