Robo3D/Marlin #22: [FR] Enable CASE_LIGHT_PIN

WheresWaldo, 2017-11-29:
### Feature Request

Please consider a slight modification to enable case lighting under g-code control. only two lines are necessary to provide support and uncommenting the define would allow direct control of additional LED lights added by the end user with appropriate hardware without the need to download firmware, recompiling and flashing firmware.

The two edits required are first in pins_ROBOMB.h &/or pins.ROBOC2.h
`#define CASE_LIGHT_PIN	44  // MUST BE HARDWARE PWM (2, 3, 4, 5, 6, 7, 8, 9, 10, 44, 45) `

In Configuration_adv.h the following change is required, uncomment the line
`#define CASE_LIGHT_ENABLE`

The end result is that you have control over a single color LED strip (user provided hardware). Control included turning the LED strip(s) on or off as well as brightness control. I have outlined all the required hardware in this Projects thread on the Robo forum, http://community.robo3d.com/index.php?threads/g-code-controlled-case-lighting.19871/

AllenMcAfee, 2017-12-12:
@WheresWaldo I have enabled this in the latest firmware.  I pulled the definition out of Config_adv and placed it in the pins doc for ROBOMB.  We are dropping the C2 pins list and consolidating to the one ROBOMB pin file. 

I am enabling this on pin 5 to start since ROBOMBs going forward will have the LED pins PWM enabled.  The current Mainboard 2.1.5 and below have a resistor that can be removed to enable PWM to the lights and controller fan (R28 and R47 by the extruder connector).  To keep it simple we are going to list CASE_LIGHT as pin 5 in the production firmware.  You're free to handle it on any pin you want in your own flavor, of course.

WheresWaldo, 2017-12-12:
Thanks Allen, I already saw this in the source and saw that you are combining the pins file. That is smart. II also saw the drop drown resisters on the main board and I will be removing at least the FAN2 to give control of the lighting. I will also write up this mod for the forum so others can modify the 2.1.5 boards if the want. The cooling fan on Pin 6 doesn't really need PWM control so I will leave it. I am going to close this issue.

closed 2017-12-12 by WheresWaldo