Robo3D/Marlin #23: DAC does not enable due to an undefined pin in Arduino Libraries

Ximidar, 2017-11-30:
Pin PJ6 on the mega 2560 is not defined in the arduino libraries for it. It does not have a pin address, so it cannot be reached by using arduino's default code (digital read/write). https://www.arduino.cc/en/Hacking/PinMapping2560

To overcome this I'm attempting to use Port Manipulation ( https://www.arduino.cc/en/Reference/PortManipulation )

In `stepper_dac.cpp` in the `dac_init` function I changed the function to use port manipulation to set the unmapped pin to low. 
```
int dac_init() {
    #if PIN_EXISTS(DAC_DISABLE) //Undefined this variable in pins_ROBOMB.h
       OUT_WRITE(DAC_DISABLE_PIN, LOW);  // set pin low to enable DAC
    #else
      DDRJ = DDRJ | B00000010; //sets unmapped pin PJ6 to an output and low
      PORTJ = PORTJ | B00000000;
    #endif
```

I'm just documenting this for now, My port manipulation may be completely off since I've never had to do this before.

I would rather not do this, but it does provide us a quick test to see if the enable pin is the issue with the DAC or not.

My next move will be to alter the mega 2560 pins_Arduino.h file to define the missing pins we are using.

AllenMcAfee, 2017-11-30:
After testing the above pin work did not have an effect on enabling the DAC.  We will incorporate the fix above for PJ6 on the ATMEGA 2560 to enable the EE_READY pin per the schematic.  EEPROM has been verified that it can be read and written to on the DAC.  

At this point the question is why the output registers are not enabled.

Ximidar, 2017-11-30:
After testing the VREF test points to the stepper drivers we have concluded that the DAC is working properly, The VREF changes properly and we can set the DAC EEPROM properly. The Issue may be with the stepper configuration itself.

AllenMcAfee, 2017-12-11:
DAC is working properly - there was an issue with the stepper driver circuit (specifically the cap between the charge pumps was incorrect).

Closing this out.

closed 2017-12-11 by AllenMcAfee