Robo3D/Marlin #24: DAC serial display and EEPROM values need to match actual channels on Mainboard 2.1.8

AllenMcAfee, 2017-12-12:
### Bug Report
- Description: The 909 command via terminal is displaying the following data:
`
Recv: echo:Stepper current values in % (Amps):
Recv: echo: X:59.99 (-20480.00) Y:59.99 (-20480.00) Z:59.99 (-20480.00) E:59.99 (-20480.00)
`
The steppers themselves need to be relabled in Marlin in two ways:

1. The dac_print_values function needs to be defined in the following way:
`  void dac_print_values() {
    if (!dac_present) return;

    SERIAL_ECHO_START();
    SERIAL_ECHOLNPGM("Stepper current values in % (Amps):");
    SERIAL_ECHO_START();
    SERIAL_ECHOPAIR(" XY:",  dac_perc(XY_AXIS));
    SERIAL_ECHOPAIR(" (",   dac_amps(XY_AXIS));
    SERIAL_ECHOPAIR(") Z:", dac_perc(Z_AXIS));
    SERIAL_ECHOPAIR(" (",   dac_amps(Z_AXIS));
    SERIAL_ECHOPAIR(") E0:", dac_perc(E_AXIS));
    SERIAL_ECHOPAIR(" (",   dac_amps(E_AXIS));
    SERIAL_ECHOPAIR(") E1:", dac_perc(E1_AXIS));
    SERIAL_ECHOPAIR(" (",   dac_amps(E1_AXIS));
    SERIAL_ECHOLN(")");
  }
`
2. The input function of the DAC EEPROM values also needs to be changed. (where EEPROM values to DAC are changes (M907)).  The current commands works as such:
```M907 X60 Y60 Z60 E60```
It will need to read something like this:
```M907 XY60 Z60 E060 E160```
or this if the integers for the extruders are confusing:
```M907 XY60 Z60 A60 B60```
We can discuss this last bit more internally to come up with a consensus.

Ximidar, 2017-12-15:
Unfortunately we cannot use two characters for an identifier in these values, However I think I have made a nice compromise. Instead of using XY, Z, E0, and E1, I've Used A,B,C,D. This makes unique channels that are easy to keep track of. I've also added these to the M909 printout. That looks like this now:
```
  void dac_print_values() {
    if (!dac_present) return;

    SERIAL_ECHO_START();
    SERIAL_ECHOLNPGM("Stepper current values in % (Amps):");
    SERIAL_ECHO_START();
    SERIAL_ECHOPAIR(" (A)XY:",  dac_perc(XY_DAC));
    SERIAL_ECHOPAIR(" (",   dac_amps(XY_DAC));
    SERIAL_ECHOPAIR(") (B)Z:", dac_perc(Z_DAC));
    SERIAL_ECHOPAIR(" (",   dac_amps(Z_DAC));
    SERIAL_ECHOPAIR(") (C)E0:", dac_perc(E0_DAC));
    SERIAL_ECHOPAIR(" (",   dac_amps(E0_DAC));
    SERIAL_ECHOPAIR(") (D)E1:", dac_perc(E1_DAC));
    SERIAL_ECHOPAIR(" (",   dac_amps(E1_DAC));
    SERIAL_ECHOLN(")");
  }
```

Then I Changed the M907 to recieve the arguments  A,B,C,D, Then S was already defined.
```
const char dac_codes[ABCD] = {'A', 'B', 'C', 'D'};
```
```
  /*
  This will now work off of channels ABCD. So M907 A100 will change the A channel on the DAC which is the XY Channel.
  */
  #if ENABLED(DAC_STEPPER_CURRENT)
    if (parser.seen('S')) {
      const float dac_percent = parser.value_float();
      for (uint8_t i = 0; i <= 4; i++) dac_current_percent(i, dac_percent);
    }
    LOOP_DAC(i) if (parser.seen(dac_codes[i])) dac_current_percent(i, parser.value_float());
  #endif
```

@AllenMcAfee I haven't pushed this yet, however it is running on Floating-Points. Please leave some feedback

Ximidar, 2017-12-15:
I added these edits to the branch: https://github.com/Robo3D/Marlin/tree/DAC_Edit

AllenMcAfee, 2018-02-13:
Fixed and rolled into RC branch

closed 2018-02-13 by AllenMcAfee