Robo3D/Marlin #30: Current code additions for G35 break UBL and MESH

WheresWaldo, 2018-02-09:
### Compile Error
When I compile with `MESH` or 'UBL' I get an error:
Compilation fails when you disable ABL and enable either MESH or UBL. Same error for both features.

Paste_the_error_text_here
```Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\Marlin_main.cpp: In function 'void gcode_G35()':

Marlin_main.cpp:5368: error: 'LEFT_PROBE_BED_POSITION' was not declared in this scope

     const float rnx = LEFT_PROBE_BED_POSITION - (X_PROBE_OFFSET_FROM_EXTRUDER), rny = FRONT_PROBE_BED_POSITION - (Y_PROBE_OFFSET_FROM_EXTRUDER);

                       ^

In file included from sketch\Marlin_main.cpp:250:0:

Marlin_main.cpp:5375: error: 'rny' was not declared in this scope

     current_position[Y_AXIS] = LOGICAL_Y_POSITION(rny);

                                                   ^

sketch\Marlin.h:250:40: note: in definition of macro 'NATIVE_TO_LOGICAL'

 #define NATIVE_TO_LOGICAL(POS, AXIS) ((POS) + WORKSPACE_OFFSET(AXIS))

                                        ^

sketch\Marlin_main.cpp:5375:32: note: in expansion of macro 'LOGICAL_Y_POSITION'

     current_position[Y_AXIS] = LOGICAL_Y_POSITION(rny);

                                ^

exit status 1
'LEFT_PROBE_BED_POSITION' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

------------------------------------------------------------
### Bug Report
- Description: Robo code breaks existing functionality
- Expected behavior: Do no harm
- Actual behavior: Compilation fails
- Steps to reproduce:
  - Do this, just disable ABL and enable either MESH or UBL
  - Do that, compile

Ximidar, 2018-02-10:
With MESH you shouldn't need a probe. The firmware is failing to compile because you still have the probe activated with a leveling algorithm that has no use of a probe. Disable these lines in the configuration.h to get it to compile. This will disable the `HAS_BED_PROBE` Constant which will prevent G35 and G36 from being compiled with the rest of the program.
```
#define Z_MIN_PROBE_REPEATABILITY_TEST //line 798
#define FIX_MOUNTED_PROBE // line 703
```
I'm not sure if ABL requires a probe or not since I have never tested or used it. However since both of these leveling algorithms are supposed to be manually set meshes I think you could get away with disabling it there as well. If not, then G35 and G36 reside in the `Marlin_main.cpp` file on line 5348. You could just comment out everything inside the G35 and G36 functions since that fix would not help with ABL or MESH leveling.

Commenting out these lines will get the firmware to compile. However the point of using Robo's version of marlin is to experience our tailored version of the firmware for Robo's printers. If you would like to use UBL or Mesh leveling You would probably have better luck using Marlin's version of their Firmware, then applying our pinout to that firmware. Personally I have not tested UBL or Mesh leveling with Robo so I cannot tell you that it will work as intended. We do not have any plans to support UBL or MESH leveling yet, so feel free to activate it, but it will probably break some functionality. Using anything besides our configuration will most likely negatively impact user experience on RoboLCD as well.  

However activating linear or bilinear leveling has been working out well in our testing. Good luck with your modding.

AllenMcAfee, 2018-02-10:
@WheresWaldo I have updated G35 to include handling the probe points for UBL and 3 point ABL, so it's compiling correctly now.  These changes were just uploaded to the RC-bugfix branch.  I did not include anything for MESH, so G35 is now wrapped in an IF statement that changes the probe points based on the leveling selected.

I also changed out the physical probe points themselves, so everything is in the bounds of the probe's reach.

Let me know if G35 helps at all with UBL.

WheresWaldo, 2018-02-10:
@Ximidar While MESH can be used without a probe it was it works and compiles correctly with both of those lines enabled in Generic Marlin. Think of UBL as sort of Automated MESH. Just so you know, I broke RoboLCD a long time ago. I have been using UBL on the R2 since Marlin version 1.1.6 and it is significantly better than the first release of Robo's Marlin.  If you go to my GITHUB you will find Marlin 1.1.8 properly configured for the R2 that compiles and works flawlessly. The only time I visit my printer is to remove a print, so I almost never use RoboLCD. @AllenMcAfee has mentioned to me that there are some fixes in the new test version of RoboLCD which I was hoping to get to today but ran into other non-3D printing related life.

Just so you know I have been the person on the forums that maintains the Latest Marlin both supporting MESH and UBL for the R1 series and the R2 series. I think I have a pretty good grasp on how Marlin works.

Remember regression testing will help you do no harm.

@AllenMcAfee I will try out G35 with UBL. Just so you know. I leveled my bed with Marlin 1.1.6 and now I am 10 of flashes later and two versions newer and did not have to re-level the bed since. As long as the EEPROM schema revision does not change the EEPROM setting are preserved in a new flash. Personally I would ignore MESH entirely, the Marlin developers wished it was gone, putting all their eggs in the UBL basket (MESH on steroids) The have been afraid to deprecate the code completely, fearing community backlash.

I am still skeptical of how the code written for G35 actually accounts for ambient light, My own opinion is that there are too many assumptions being made of what is actually being measured. I am also thinking about experimenting with the MULTIPLE_PROBING variable which replaced PROBE_DOUBLE_TOUCH in the latest Marlin would not do the same thing with a value greater than 2.

WheresWaldo, 2018-02-10:
Ideally, or maybe it is just me. If you could build a wizard that walks through UBL then an R2 or C2 user might never have to level again, in normal use. There is an easy menu driven system if you are using a smart LCD controller, especially after I pushed Roxy_3D really hard on why use UBL when MESH worked so well. Now that UBL is much more mature, it is closer to a one and done type of leveling than any ABL method previously used. You can also save multiple meshes in EEPROM if you are a frequent bed changer.

AllenMcAfee, 2018-02-13:
@WheresWaldo we'll add this to the roadmap for supporting UBL in an upcoming version of RoboLCD.  When we revisit it we'll update it in a new issue to list out the scope of the change and take public feedback.  For now, we can close this issue out since UBL and Mesh are compiling after @Ximidar gives it a pass over to check my work.  @WheresWaldo do you agree?

WheresWaldo, 2018-02-13:
Agreed and closed

closed 2018-02-13 by WheresWaldo